ckflight/ESC-BLDC-HARDWARE #1: Is there any info about the max Voltage this can handle?

MxFrs, 2023-02-11:


ckflight, 2023-02-11:
If you can check the schematic pdf file you will see that l78l12 regulator is used which is capable of handling input voltage up to 35 V volts.

closed 2023-02-11 by ckflight

catphish, 2023-05-06:
I believe the specified MOSFETs are rated for an absolute maximum of 30V Vds. This means that overall, as designed, the maximum practical voltage is going to be 30V. Given that this design is likely to be used to switch voltage from lithium batteries, and allowing a margin of safety, it looks like it should be good for up to 6S (25.2V at 100% SoC).

With regard to minimum voltage, if I'm understanding correctly, the driver requires an absolute minimum of 5.4V, and the 12V regulator has a dropout voltage of 1.7V, so the absolutely minimum operating voltage would be 7.1V. Again, assuming lithium batteries and a small margin of safety, this would equate to a minimum of 3S (9V at 0% SoC).

I hope you don't mind me resurrecting this thread. I thought it might be worth mentioning these conditions in the README.