ckflight/ESC-BLDC-HARDWARE #2: Motor not spinning properly

MxFrs, 2023-05-21:
Hello, I have rebuilt your design with some slightly different components to raise the maximum voltage. I managed to get the motor moving in an uncontrolled fashion. It looks like desync to me, but not really sure. I measured a lot of signals with an oscilloscope and for me all signals look pretty much fine, but as you can see in the video, it doesn't spin the motor. Did you ever experience something similar while developing/testing your design? 
I've also attached my KiCad schemtic so you can maybe check weather I made any obvious mistakes (I tried to copy your design as much as possible).

https://github.com/ckflight/ESC_BLDC_HARDWARE/assets/35507432/e0968a6a-ea30-4bd2-8eca-dfb2842b4fb6

[image]

ckflight, 2023-05-22:
Hi with my design i did not experience anything like in the video. Its not
desync for sure though. It cannot commutate it seems like timing related
issue. I have check the transistor datasheet you used and its gate charge
and on off time is similar to the transistor in my design. I would suggest
changing the firmware timing try other ones like c h 20 25 etc.

Also pcb design is really important i made 4in1 esc with same setup and it
did not work properly. Meanwhile it was fine in 1 esc design.

Let me know if you figure it out the problem. I wonder what is the problem.

Regards.

MxFrs, 2023-05-22:
Thanks for the tips, I will try with the different firmware. 
Also, I forgot to mention, that I am actually using different Mosfets (https://www.mouser.de/ProductDetail/?qs=gt1LBUVyoHmr9i%252B8H6gEPw%3D%3D). They have for example a higher gate charge. Can you specify what to look out for to check, weather the gates are driven properly? I guess I would want to see a kind of square signal on the gate pin which has high enough voltage?

ckflight, 2023-05-22:
Check description of the project there are 2 formulas. Simply transistor
with higher gate charge meaning more powerful, it needs higher current at
micro milli second level like from 2 3 ampere to 0 in millisecond to
properly drive the gate of the transistor.

Addition to the suggestions i said in thebprevious mail also decrease the
gate resistor value from 47 to something lower like 30 20 etc to drive with
higher current.

ckflight, 2023-05-22:
Long story short, The transistor you use is powerful and it has higher
switching time because it needs higher current to charge itself therefore
it takes time. Try firmware with higher delay than 30 also decrease gate
resistor value little bit. It should work because it is working but not
commutating properly roght now.

MxFrs, 2023-05-23:
I tried all the following firmwares with 47 Ohm resistors and, by adding the same in parallel, 23.5 Ohm. I tried C-H-30, 15, 5, 40, 50, 70 and 90. Unfortunately no real difference between all the versions. I also lowered the startup power to 0.125 which resulted in the motor starting slower and actually making about one revolution before starting to do weird stuff again. Any other ideas?

ckflight, 2023-05-23:
Then as i mentioned before, pcb design is important. When 2 of the motor
connection is connected third one is for feedback for detecting the phase
of the motor. Maybe that signal affected by return currents etc. Can you
compare your esc design's motor phases to a working esc.

Also can you change the transistors to a less powerfull one to cancel out
that option as well.

Npidys, 2023-06-11:
@MxFrs  Please check your gate driver's maximum source/sink current capabilities; they are not enough to drive the FETs in your design, assuming the fastest turn-on. It is possible to drive a FET slower with less current, (turn-on time is proportional to the gate current of the FET), but that has a lower limit because a second FET turn-on cycle might start before the first FET is wholly discharged, due to overlapping drive signals from the FC; this would create a shoot-through (electrical short between two phases of the 3-phase half bridge) and fault the ESC.

closed 2024-02-02 by ckflight